imaginevision/Z-Camera-Doc #1: RESTful HTTP API?

pizthewiz, 2015-09-21:
Generally the API called out in [http.md](http.md) is not RESTful; `GET`s are used when deleting and adding resources - is this due to the limited HTTP verb support in the E1 hardware or would you be open to a proposal for a RESTful HTTP API?

A couple examples off the top of my head:
* `GET` _/DCIM/100MEDIA/EYED0001.MOV?act=rm_ could be a `DELETE`
* `GET` _/ctrl/session?action=quit_, _/ctrl/shutdown_, _/ctrl/reboot_, _/datetime?date={YYYY-MM-DD}&time={hh:mm:ss}_, _/ctrl/still?action={cap}_, _/ctrl/mode?action=to_pb_ … could be a `POST`
* `GET` _/ctrl/mode?action=query_ could be query-less

pizthewiz, 2015-09-22:
* `GET` _/ctrl/card_ could respond with 204 if present and 404 if not
* `POST` _/ctrl/card/format/_

benhinchley, 2016-07-13:
An RPC based api would probably be better suited to this kinda of work

paviro, 2020-07-20:
I would second this issue and would also favour a RESTful API.

milliyang, 2025-12-09:
Why the Non-RESTful API Design?
This design choice is typically driven by historical compatibility, implementation simplicity, and ubiquity:

1. Maximum Compatibility (The Primary Reason)
2. Simplified Client and Server Implementation
3. Leveraging Browser Accessibility
4. The "Query Parameter as Verb" Pattern, This is actually an RPC-over-HTTP style, not a resource-oriented REST style. It focuses on "what command to execute" rather than "what operation to perform on which resource." For simple device control, this pattern is very intuitive.   (personally)

closed 2025-12-09 by milliyang