imaginevision/Z-Camera-Doc #3: Undefined camera discovery for HTTP API

pizthewiz, 2015-09-21:
How would one go about discovering a camera when using the [HTTP API](http.md)? If BLE, could you provide the services, characteristics and sample response data?

ailikeimg, 2015-09-21:
Regarding the BLE, it provide some characteristics.
Regarding HTTP API, I am not clear about what you mean.

pizthewiz, 2015-09-21:
Sorry for the confusion, if I were using the [Z Camera](http://www.z-cam.com/index.php?r=en/product/app) app, how does it find the camera to connect to when using Wi-Fi (which I assume uses the HTTP API) because it doesn't allow us to manually specify an IP.

pizthewiz, 2015-09-21:
Perhaps it uses [ZeroConf](https://en.wikipedia.org/wiki/Zero-configuration_networking), if so what is the service name?

ailikeimg, 2015-09-22:
The camera is AP, so the smartphone need to connect to it directly use a fixed IP.
For ZeroConf, it's suitable for STA.

pizthewiz, 2015-09-22:
Ahh, I hadn't realized that the camera was the access point when using the HTTP API, that might be worth mentioning in [http.md](https://github.com/imaginevision/Z-Camera-Doc/blob/master/http.md).

In the future will there be any consideration for allowing the camera to join an existing network (usually via paired mobile app and BLE) so that the connecting client could have an active outgoing internet connection? In that case, ZeroConf could be a nice discovery mechanism.

pizthewiz, 2015-11-09:
How is the camera discovered via BLE, does it advertise a specific service?

ailikeimg, 2015-11-18:
Well, I will update the info about BLE

pizthewiz, 2015-11-22:
Does the E1 use [Wi-Fi Direct](https://en.wikipedia.org/wiki/Wi-Fi_Direct) instead of being a full [Access Point](https://en.wikipedia.org/wiki/Wireless_access_point)? I'm trying to sort out why clients connected to the camera cannot communicate with each other.

bragonznx, 2016-07-01:
Hi Ailikeimg

Could you provide the BLE documentation.
In fact i'm trying to find how to boot the camera via bluetooth
I did some reverse engineering, but it'll be appreciate if you can share information.

q3o, 2018-02-12:
Hi, anyone knows how I can connect the E1 to an existing network? I change to STA wifi mode and it say wifi "ivt" and connecting to some ip, but it doesnt really connect to anything nor does it give the choice to pick up an existing network or ask for password or anything?
Any help? thanks