imaginevision/Z-Camera-Doc #4: Session ownership and representation

pizthewiz, 2015-09-27:
The [http.md](https://github.com/imaginevision/Z-Camera-Doc/blob/master/http.md) doc mentions that the `/ctrl/*` interface is only accessible to the session holder but it isn't clear how session ownership works and how it is represented.
- How is a session started?
- How is the session represented, a token?
- How does a client know if there is an existing active session?

ailikeimg, 2015-10-26:
How is a session started?
>>> Every request in /ctrl/* will start a session, unless it return 409 in the status code.
If you want to give up the owner ship, you can call /ctrl/session?action=quit
And you can keep the session alive with a call /ctrl/session in every 30s

How is the session represented, a token?
>>> It's base on IP. You don't need to put it in your request.

How does a client know if there is an existing active session?
>>> If you access he /ctrl/* interface and the status code in HTTP header is 409, it means the camera is owned by other client.

pizthewiz, 2015-10-28:
Gotcha, so the session is per-contacting IP, very good to know! Might it be appropriate to add this info to the docs?

pizthewiz, 2015-11-02:
Is there a standard session timeout period?
Do all calls from the client extend the session, or only calls to `/ctrl/*`?

ailikeimg, 2015-11-03:
30s timeout.
Any call /ctrl/* is ok. e.g. /ctrl/session?hb=1