imaginevision/Z-Camera-Doc #9: Undocumented query-less mode

pizthewiz, 2015-11-04:
`GET`: `/ctrl/mode`

Is this synonymous to `/ctrl/mode?action=query`?

pizthewiz, 2015-11-09:
It seems it is synonymous to the query action, 0.16 returns `{"code":0,"desc":"","msg":"cap"}` or `rec` what have you.

ailikeimg, 2015-11-10:
cap means the camera is in still capture mode, while, rec means in movie recording mode

pizthewiz, 2015-11-12:
So a query-less `GET` to `/ctrl/mode` is synonymous to `/ctrl/mode?action=query` then?

ailikeimg, 2015-11-18:
haha, no~ do you think it will be better?

pizthewiz, 2015-11-18:
> do you think it will be better?

Sorry I'm confused, I wasn't criticizing the API but just asking what the query-less request meant since it wasn't mentioned in the documentation.

pizthewiz, 2015-11-22:
My goal is to help generate _complete_ documentation for the HTTP API and so far I have found that the [doc](https://github.com/imaginevision/Z-Camera-Doc/blob/master/http.md) leaves out a lot of detail.