imaginevision/Z-Camera-Doc #11: Undefined gallery video info response properties

pizthewiz, 2015-11-10:
Well, `w` is width in pixels and `h` is height in pixels but I'm unsure what `vts` and `vcnt` are (encoder settings?) and `dur` seems to be duration in some strange units 10^-5 seconds?
```
http GET "10.98.32.1/DCIM/100MEDIA/EYED9853.MOV?act=info"
HTTP/1.1 200 Ok
Cache-Control: no-cache
Connection: keep-alive
Content-Length: 83

{"code":0,"desc":"","msg":"","w":3840,"h":2160,"vts":90000,"vcnt":256,"dur":768768}
```

ailikeimg, 2015-11-10:
forget about the vts/vcnt:)
duration is counter base on timescale.

pizthewiz, 2015-11-12:
`vts` is Video TimeScale then? If so, we'd need it to be able to interpret `duration` no?

ailikeimg, 2015-11-18:
Yes, you are right. duration =  json.dur/json.vts