imaginevision/Z-Camera-Doc #12: I/O Connector

francisrath, 2015-11-23:
Hello,

I have not heard back about I/O connector type or pinout. Can you please put that here as well?

ailikeimg, 2015-12-01:
Please check the latest document.

danellecline, 2015-12-02:
I have the same question.  Is USB-LG12-0245.pdf the latest document you are referring to ?

ailikeimg, 2015-12-02:
Yes.

danellecline, 2015-12-02:
Thank you very much for your prompt response.   Could you please tell me a little more about the manufacturer/part number of that connector, and the pinout, for example, where is VCOM pin 1, SSIO EN5/UART TX pin 6, and ground pin 12 ?  I'm having difficulty finding information on that part.

ailikeimg, 2015-12-08:
We have a file named pin.jpg in the folder

danellecline, 2015-12-08:
Thank you. Yes, I see that file. What I don't understand is where physically pin 1 is on the connector described in the drawing USB-LG12-0245.pdf which looks like a micro usb connector.  I'm in the US. Could you possibly share where you found that part for us who need to interface to it ?

s3erjaeh, 2016-03-22:
Hey,

I got the I/O Connector Plug, solder it to a breakout-board and try to get a connection to the Camera. The github documentation says, the camera will look at i2c address 0xa3 to read a identification byte to identify the connector type. I simulate the EEPROM with a Arduino bord, but I can't get a read request from the Camera.

Also the CVBS output is not working and in the camera menu the CVBS output option is disabled. Is this depending on the current Firmware 0.26 and will be fixed in the next versions, or do I have some mistakes in my setup?

tiko0071, 2017-03-28:
hi, does anyone knows how to activate cvbs output? thanks

francisrath, 2017-04-11:
Well, this crap is apparently abandoned... :(

closed 2017-04-11 by francisrath