imaginevision/Z-Camera-Doc #13: IO connector where to buy?

Smuedl, 2015-12-08:
Hello,

can somebody tell me where I can buy this plug? 
My location is germany.
Maybe someone knows the name of this connector.

Thank you very much.

danellecline, 2015-12-08:
I have the same question.  I'm in the United States.  Thank you.

danellecline, 2015-12-15:
Can someone please respond to this question. Thank you very much in advance.
[image]

Smuedl, 2015-12-15:
Hello,

after one week searching I think I found the plug:

http://de.aliexpress.com/item/20-Pcs-Per-Lot-Mini-USB-12-Pin-Male-Plug-Connector-For-DIY/32527483366.html?ws_ab_test=searchweb201556_4,searchweb201644_0_79_78_77_82_80_62,searchweb201560_4

I hope it will work!!!

best regards,

Stefan

sinjens, 2017-11-13:
I tried this plug https://www.ebay.com/itm/180396604436

And it was not the correct one.  It really bugs me that Z-cam people can't comment and let us know which plug is actually the correct one.  The camera has some potential but you sure as hell don't get much help from the manufacturer :(

RagnarJensen, 2018-02-17:
I have found the plug on eBay.
https://www.ebay.co.uk/itm/8Pcs-Mini-USB-12-Pin-Male-Plug-Connector-For-DIY/170580271467

I'm currently experimenting with CVBS and getting a wired remote going
[image]

Here's the pin-out, with the numbering scheme of the [pin.jpg](https://github.com/imaginevision/Z-Camera-Doc/blob/master/expansion/pin.jpg) file:
[image]

Cheers,
Ragnar

sinjens, 2018-03-26:
Thanks for the tip on the plug Ragnar.  I got a plug now :)  (although I have no idea what it's called).  Did you get your wired remote going ?

Chenel000, 2018-04-10:
Thank you Ragnar for the information provide!

I am trying to buy the connector, but the Ebay link has ended, and amazon.de one is not available. 

Any other link to buy the connector please?, I am at Spain, but any provider as Aliexpress, Amazon, Ebay ,... is wellcome.

RagnarJensen, 2018-04-10:
It looks like this one is the same.
https://www.ebay.co.uk/itm/12-x-Mini-USB-Plug-Male-12-Pin-Used-for-SAMSUNG-Product/180396604436?hash=item2a0079bc14:g:OH8AAOSwyTZUXbsd

/Ragnar

RagnarJensen, 2018-04-10:
@Chenel000
It's all soldered up, but I'm looking for some neat little box to put it in.
I'm thinking of something that will attach to the tripod hole underneath the camera body.
Maybe I will persuade myself to finally get a 3D-printer :wink:

Chenel000, 2018-04-10:
Thank you for your fast response Ragnar!

I have checked the link but, while the advertisement states 12 pin, the picture of the connector seems to have 14 pins, see picture attached.

[image]

Chenel000, 2018-04-10:
Yes, I recommend you a 3D printer, I use one for the hardware developments and it is a significant step ahead.

Chenel000, 2018-04-10:
This one seems to have 12 pin. Do you think it fits?
https://www.ebay.co.uk/itm/12-Mini-USB-Plug-Male-12-Pin-Used-for-LG-Digit-Product/180396604315?_trkparms=aid%3D222007%26algo%3DSIM.MBE%26ao%3D2%26asc%3D49129%26meid%3Db4f16dd4a1194a49adab75ce2aaba1a5%26pid%3D100005%26rk%3D1%26rkt%3D3%26sd%3D180396604436%26itm%3D180396604315&_trksid=p2047675.c100005.m1851

sinjens, 2018-04-10:
@Chenel000 The "for Samsung product" plug is probably wrong, that was the plug I first bought, and it did not fit, also had the same product image.  The second you link seems to have all pins on one side, so that's probably wrong also.  Seems I got the latest from Ragnars original link.

@RagnarJensen Any details on your implementation would be appreciated.  I also plan to implement a wired remote for use in model aircraft, 1.pri is to be able to take photos, 2.pri would be to be able to switch between video and photo, take photos and start/stop video.

RagnarJensen, 2018-04-10:
@sinjens  OK, so that's a no-go then for the "for Samsung product plug" :disappointed:
Yes, I'm planning a little write-up. Right now I have your 1.pri fully working. I have tested your 2.pri OK on the bench using an RS-232 interface.

@Chenel000 I have more plugs than I need. I could drop a couple in an envelope and send to you.
Is my e-mail visible to you on my profile page?

bronzeproductions, 2018-05-18:
Hi all - has anyone figured out which plug to use for the I/O port?

RagnarJensen, 2018-05-31:
I mentioned this one on April 10:
https://www.ebay.co.uk/itm/12-x-Mini-USB-Plug-Male-12-Pin-Used-for-SAMSUNG-Product/180396604436?hash=item2a0079bc14:g:OH8AAOSwyTZUXbsd

The description says 12 pins, but the pictures show 14. But, the measurements in the description looked OK, so I bought some just to have a look at them.
They do have 12 pins and if you remove the metal shield on the plug, it actually fits the camera :grinning:

[image]
Left: plug as it arrived. Middle: metal shield removed. Right: the shield itself.

I have soldered up a cable with it and it works!

/Ragnar

RagnarJensen, 2018-06-02:
Looking closer at the plug and measuring it with calipers, I thought it really should fit the camera as it is, without removing the metal shield. And it does. It is a snug fit, you have to insert it straight in. If you angle it even the smallest amount, it won't insert. So, you might still want to remove the metal shield to make it easier to insert.

[image]

/Ragnar

dr-nick, 2018-07-15:
If anyone is looking for another source, I can confirm this one fits. Haven't wired it up yet however. Took about 10 days to get to California from China. 

https://www.ebay.com/itm/180396604436

[image]

DeanZwikel, 2019-02-05:
@RagnarJensen 

Could I ask what size wire you used to solder to the connector? Also, do you know if all 12 pins of the expansion port are used when a sync cable is connected to the E1? I'm trying to mount two cameras side by side and the sync connector sticks out to far for me to get the cameras close enough together. I'm thinking of making an extension cable with a low profile right angle male connector on the end that connects to the camera and a female at the other end to accept the sync connector. I could make the cable with 12 conductors but that means a very thick cable and a lot of soldering on each end. I'm hoping not all conductors are needed. Thank you.

RagnarJensen, 2019-02-05:
I'm using this wire: https://www.electrokit.com/en/product/hook-up-wire-0-14mm2-white-10m/
It's a 1 mm thick single-conductor wire and I use only as many as needed. I bundle them up with heat-shrink tubing.

You don't need all 12. If you don't need composite video, then typically six wires are needed.
- Power (to the EEPROM)
- Ground
- SCL
- SDA
- TX (or wired remote shutter - full press)
- RX (or wired remote AF - half press)

A bundle of six conductors is about 2.5 mm in diameter.

[image]
Connector with all 12 soldered.

Could you mount one camera upside down, to have the connector on the other side? The camera has a "Rotation" setting to flip the picture.

dr-nick, 2019-02-05:
That’s a great idea!

> On Feb 5, 2019, at 11:08 AM, Ragnar Jensen <notifications@github.com> wrote:
> 
> Could you mount one camera upside down, to have the connector on the other side? The camera has a "Rotation" setting to flip the picture.

DeanZwikel, 2019-02-05:
> Could you mount one camera upside down, to have the connector on the other side? The camera has a "Rotation" setting to flip the picture.

Thanks. That's an interesting idea. I might be able to do it. It does complicate things. I would need a special mount to support the upside down camera and the control buttons would also be upside down. That would make it somewhat confusing to change settings and, for photos, I need to hit the up arrow button on each camera to trigger AF prior to hitting the down button on the master to trigger the shutter. That could cause mistakes while shooting. All that said, it's still an idea worth exploring. Thanks!

Thanks for the info on the wire and the pins.

DeanZwikel, 2019-02-05:
Actually, I see that using "Rotation" to flip the picture also flips the functions of the arrow buttons so that's a non issue.

RagnarJensen, 2019-02-05:
@DeanZwikel 
From your mentioning a "sync cable" and "master"/slave" I get a feeling you are doing something I have not tried myself. Are you connecting a sync cable between the two cameras and nothing else? Push the shutter button on one camera and both fires?
Then my wiring might not be the best, e.g. I wouldn't connect the two camera's power lines together.
What is this mysterious sync cable?

DeanZwikel, 2019-02-05:
@RagnarJensen 

> @DeanZwikel
> From your mentioning a "sync cable" and "master"/slave" I get a feeling you are doing something I have not tried myself. Are you connecting a sync cable between the two cameras and nothing else? Push the shutter button on one camera and both fires?

Yes.

> Then my wiring might not be the best, e.g. I wouldn't connect the two camera's power lines together.
> What is this mysterious sync cable?

I am using the sync cable that is part of the iZugar Z2XL180 kit. See diagram and the link below.

[image]

http://www.izugar.com/shop/product/z2xl180/

RagnarJensen, 2019-02-05:
Hmm, so it's a dongle in each camera and the dongles are wired together. Then it's a different ballgame all together. I have no idea which pins of the I/O port the dongle uses.

I notice that the cameras are in portrait orientation in the picture on iZugar's site...

DeanZwikel, 2019-02-05:
> Hmm, so it's a dongle in each camera and the dongles are wired together. Then it's a different ballgame all together. I have no idea which pins of the I/O port the dongle uses.
> 
> I notice that the cameras are in portrait orientation in the picture on iZugar's site...

Yes. I'm using the cameras in landscape orientation to capture 3D/stereo with normal rectilinear lenses.

RagnarJensen, 2019-02-05:
Yes, I think so too.