imaginevision/Z-Camera-Doc #14: About Streaming

Acoh, 2015-12-09:
Would you elaborate how one can use streaming?

onomonomedia, 2017-07-04:
I'd like  to second this request.  It seems like it was there before but now it's missing from the documentation

benhinchley, 2017-12-17:
From the docs previously
```
A streaming service is provided over port 9876, via a TCP based server.

To fetch a single frame:

send 0x01 to the server to request a frame
recieve data (4 byte payload len + following the frame data)
Repeating that method to get continuous frames.

Data format in payload:
Each payload of the data is H.264 encoded data. If the fps is 30, you can get 30 H.264 data in one second.
```

gfx-alex, 2021-12-17:
Hello, I didn't find the best place for asking, but does z cam e2 support streaming 1920x180/120 fps via NDI or like virtual cam?