imaginevision/Z-Camera-Doc #18: auto start

stephanschulz, 2017-03-03:
i was wondering if there is any option for the E1 to start automatically once it received power from the power supply.

we are hoping to include the camera in a permanent installation and want the camera to shutdown and restart along with the computer that received live video over the HDMI output.

thanks for any advice.

benhinchley, 2017-03-17:
I have tried to do a similar thing, but never found a working solution. The best solution I could think of was hijacking the BluetoothLE wake from sleep functionality to trigger the camera to start up again, but this does require setting up a BluetoothLE module to talk to the E1 when it is in sleep mode.

stephanschulz, 2017-03-17:
thanks for your reply.
my mac book pro has build in bluetooth. do you think that could be used to talk to the camera?
i am not super familiar with the different types of bluetooth and don't know if LE is comparable with 3.0, etc.
I also have this unit
https://www.adafruit.com/products/2649

thanks a lot.

stephanschulz, 2017-03-17:
it seems running
system_profiler -detailLevel full SPBluetoothDataType | grep "LMP Version"
in the terminal of macOS will tell you if you have LE or not.
found it here:
https://superuser.com/questions/702670/how-can-i-tell-if-my-mac-has-bluetooth-4-0-low-energy

lupupitkanen, 2017-10-23:
Have you found any solution not involving Bluetooth @stephanschulz?

The E1 camera would be otherwise perfect for my use case but unusable without auto power on.

Buy the way I haven't even been able to remote power on the the E1 using the iPhone App and Bluetooth. It seems to begin the boot-up process but never finishes (firmware version 0.30 and app version 1.8).

stephanschulz, 2017-10-23:
We made this C++ library for www.openframeworks.cc  to communicate to the camera over wifi.
https://github.com/antimodular/ofxZCamE1

But the camera always needed to be powered on manually.

We also got a ZCam C1 model, which has a USB-C input. And here you can easily connect an external circuit to power the camera on/off. I'm attaching the docs they sent me.

[image]
[image]

lupupitkanen, 2017-10-23:
Thanks @stephanschulz. I've never heard about the C1 before, need to check that out.

Although for my needs a simple firware option to power on the camera automatically when receiving DC power (like stated in your first post) would be enough. I don't need to remote control the actual camera.

stephanschulz, 2017-10-23:
true. a simple firmware update would be nice. but I have not found one yet.

qhulamm, 2017-11-07:
@benhinchley Were you successful in getting the bluetoothLE wake functioning properly?  Could you share a code snippet that worked?  I have an array of cameras and it's annoying have to power each one on everytime power is off.  I'm trying to create a power on script in Windows.

skadiexe, 2018-04-02:
Team Z-CAM tell me thats there is no option to autostart camera due to hardware limitations.

erric0x5b, 2018-04-03:
Hi, I'm also trying to find a way to auto-start the Z-cam E1. it could be possible for me also thru BLE. 
Does anybody found a way for it? Can the Z-CAM team please explain the BLE protocol?
thank you for any advice.

tbardenvt, 2020-10-21:
We are using Z Cam in our performing arts school as part of a multi-cam streaming / recording solution for performances. Also, to do what we call "Super Zoom" training.

Because we've gone with an all NDI transport schema I use our POE managed switches to control power.

falcaosurfline, 2021-03-10:
Any update on the auto-power-on feature?

tbardenvt, 2021-03-10:
If you power the ZCAM from a POE port on a network switch it becomes a question of the degree of control enabled by the switch.

The ones we use are Netgear (GC108PP) Smart Managed switches. These can control POE by port and also power off and on automatically on schedule. I don't know for sure but I imagine one can use an API to software control them too.

The E2Cs we have "turn on" when POE is supplied after any interruption.