imaginevision/Z-Camera-Doc #19: what is USB RNDIS ip address?

stephanschulz, 2017-03-03:
i'm trying to use the USB network.

which IP do i have to set my computer to in order to communicate with the camera over USB?
i tried the 10.98.32.3 that i have when using wifi, but it's not answering any of my pings

benhinchley, 2017-03-17:
The USB network feature was not fully implemented in the E1 from what I found. Though the correct IP address is `192.168.168.1`, to connect to it via usb you will need to connect to the `USB0` networking device and set an IP Address in the right range, the easiest way to do this is on a linux machine.

francisrath, 2018-01-07:
Is it possible to control the camera at all over USB with the current firmware? I would like to use it in a time lapse setup, starting/stopping the sequence, and moving pictures from the camera to a PC (Raspberry PI for example).

rlamarche, 2018-02-22:
Hi,
I successfully connected to the zcamera through usb port in networking mode on IP 192.168.168.1.
Thanks for the info on the IP Address.

I have also successfuly got the liveview in h264 and displaying it in a Qt application.

But I find that the resolution is very low (less than 640x480). I wonder if it is possible to increase it.

Not that the camera is also responding on the http api through the usb port.

rlamarche, 2018-02-23:
Thanks. I looked but there is no http control of this kind.
But : when changing the recording format in the camera it self, the h264 resolution changes.

in 4K, I get the lowest resolution. But when choosing 2160 (4/3), I get 640x480.

Summary :

I get 432x240 in 4K mode.
I get 640x480 in all 4/3 modes.
I get 640x640 in all 1:1 modes.
I get 848x480 in 1080, 720 and WVGA modes.

So the best mode for both recording and "liveviewing" on remote is 2160.

WiFi or USB Network mode have the same behaviour. There is just a different IP.
And when I enable USB Network mode, WiFi mode is no longer available.

rlamarche, 2018-02-24:
Here you can find the code of my test :

lauramorillo, 2018-03-08:
@francisrath did you manage to connect the camera to a Raspberry Pi using USB Network?

francisrath, 2018-03-08:
No, I gave up. Not putting any more effort into this camera as it seems like they didn't bother following up on their promises :(

Tried to contact them regarding a new camera, but got no reply there either...

lauramorillo, 2018-03-10:
@francisrath if you still want to do something with it I managed to connect. The main problem was the cable I was using! If you enable the usb network you will only need to modify /etc/dhcpcd.conf to add:

`interface usb0`
`static ip_address=192.168.168.2`
`static routers=192.168.168.1`
`static domain_name_servers=192.168.168.1`

And then you should be able to connect to the http api calling http://192.168.168.1

It doesn't work with all the usb cables, so you should first ensure that the raspberry pi is recognising a device interface usb0.

eungyoda, 2018-03-16:
Can I change the ip address of the E1 camera? I want to connect and control several E1s.

jlucidar, 2018-04-05:
telnet port is open, so you can reconfigure the whole thing to your needs.

eungyoda, 2018-04-06:
@jlucidar  Oh! cool! Can you tell me the root account information?

rlamarche, 2018-04-06:
you mean that we can connect to the camera with telnet ?

eungyoda, 2018-04-06:
i logged in as the "root" account. thx @jlucidar 
@rlamarche yes! we can connect to the camera.

[image]

rlamarche, 2018-04-06:
mmh that's interesting. Do you think that we could unlock or add some features to the camera by this way ? Or is this an isolated environment ..

rlamarche, 2018-04-06:
I did connect to the camera, thank you for the tip.
Now we should be able to do some hacks by patching the ambarella firmware no ?
See here : http://konradit.github.io/hacking.html

eungyoda, 2018-04-10:
I got the right to read and write. But I made the wrong bashrc file, and E1 became a black brick.
If you want to change the write permission, try as follows. Caution. It can be black brick.
1. connect telnet 192.168.168.1
2. account : root
3. mount -o remount,rw devtempfs /

jey18, 2018-07-02:
Ok i found a way to unbrick using a autoexec.ash script.
That was painful :)

jey18, 2018-08-15:
I dont have a computer to read my sd card because i'm on holidays. I think if you copy the firmware on sd card root, then create a autoexec.ash with :
firmfl prog c:\E1fw.bin
Then the camera will flash a valid firmware and boot again. You must eject your sd card as soon as you read 'update complete' or something like that on the small LCD display. If not, the flash process will start again.

vvjuss, 2018-08-15:
the camera turns off after 10 seconds. and I think autoexec not started.

jey18, 2018-08-16:
Did you edit it in unix text format ? If you are on Windows with notepad++.  If not , it will be ignored.

vvjuss, 2018-08-16:
Im using nano on macos or ubuntu. But nothing. Also I tried simple commands like ps > /tmp/fuse_d/log.txt, but it seems that the autoexec is simply not executed.

jey18, 2018-08-16:
Try:
############
dmesg rtos on
sleep 10
dmesg rtos -3000 > c:\log.txt

############

And see if log.txt  is created.

vvjuss, 2018-08-16:
You're right! It's all in the last line. I opened it in sublime and fixed. Will try to flash firmware.
Thank you!!!

chandranpranav, 2020-10-02:
Heyy guys I would really like some input here, I'm trying to connect my GoPro Hero 8 and use it as a camera on OpenCV applications. I'm using multiple cameras for object detection.  When I connect my GoPro it is detected as a Network Adapter under the name 'GoPro RNDIS Device' I can use it on Skype, Zoom, and other applications but when I try to connect two cameras it detects only one camera.. But in device manager detects as two. What I want is to read these camera inputs through OpenCV. Any input would be appreciated.