imaginevision/Z-Camera-Doc #20: UART control

tiko0071, 2017-03-28:
Hi, i'm trying to control the camera via UART but i think i forgot something, because camera doesn't do anything
for example : 
terminal 115200 bauds 8N1
to start record, i send  0x03 0xea 0x02 0x01 0x05 , and nothing happens.
 what's my mistake?
Thanks

garciaanto, 2017-12-12:
Hi Tiko,
I am also trying to enable the UART mode.
I am doing whtat the doc is saying but is not working for me.
The address that the cam lookfor the eeprom is different to the one in the documentation...

Any idea? somebody have enabled the UART modE?

Thanks to all in advance

tiko0071, 2017-12-12:
I just bought the stuff to test it today. I'll tell you when it will works. Or i f someone succeed in uart communication, please share. Thanks

-------- Message d'origine --------

skadiexe, 2018-01-11:
external eeprom programmed and connected, but there is no activity on i2c, even then the camera powering on. Always logic 1 all time on SDA/SCL