imaginevision/Z-Camera-Doc #22: Enabling UART mode

garciaanto, 2017-12-12:
Hi,
In the documentation is shown that address 0xa3 for the external eeprom must be used.
I have implemented a kind of "sniffer" for the i2c and what I see is different:

    - While booting the cam try to read the address 0x51 1 time
    - After that is in a loop asking to read 0x48 

If I answer with the Byte 0x03 (uart mode) in the first query nothing is happens
If I answer with 0x03 to all the queries the cam do not start, so its power on but nothing in the display (frame, layout...)

Any help?

skadiexe, 2018-01-11:
0x51 is 0xA3 shifted 1 bit right. it is the same 7-bit adress. 
1 or 0 in LSB means Read or Write while accessing eeprom

p.s. my camera do not have any activity on i2c overall, always logic 1 on SDA/SCL

RagnarJensen, 2018-02-17:
I have it working on the bench, using a real EEPROM at address 0x51.

When the camera starts up, the first transaction on the i2c bus my logic analyzer sees is this:
[image]
It reads the first byte from device 0x51,  which returns the value 3.

Then the camera enters a loop, continually reading from device 0x48, which is a temperature sensor inside the camera: 
[image]

It looks like a rat's nest, but it works... :-)
The terminal program on my PC has macro key functionality, which enables me to send binary strings to the camera.
[image]

[image]

Cheers,
Ragnar

CGFlyer85, 2018-04-22:
@Ragnar Jensen, that wonderful ! Is there any way we can talk about all of this, and if possible if you can help me make it work my side, off course if you need something in compensation (wink, wink) I'm okay with that's.

garciaanto, 2018-04-22:
I totally forgot this post that I opened. I have it also working.
In my case I couldnt make it works with eeprom but I used an small arduino instead. And to be honest I do not respond to the temperature query :)

To test it I just used a simple vb .net program to test some functions and more or less was working.
Also, I upgraded the cam to the WEB latest on using the manual process (just drop the image on the root folder)

Good luck!

garciaanto, 2018-04-23:
Sure:

```
#include <Wire.h>

void setup() {
  Serial.begin (9600);
  Wire.begin(0x51);
  Wire.onRequest(requestEvent);
}

void loop() {
  // :DD
}

void requestEvent() {
  Wire.write(0x03);
  Serial.println (" envío un 0x3");
}
```

CGFlyer85, 2018-04-23:
@garciaanto do you have a wiring Shem for the Arduino ?

CGFlyer85, 2018-04-23:
Is it even possible to do this ? Because that really one thing that I need, be able to manually focus thought the I/O port...

CGFlyer85, 2018-04-24:
@garciaanto did you find a way to manually focus with it ? Don't seem to find the code to use...

RagnarJensen, 2018-04-25:
Yes, it is possible to focus "manually" with UART commands. I put "manually" in quotes, because the camera must be in AF mode for it to work.

I've been experimenting with some of the commands in "Z-Camera-Doc/expansion/expansion.md" and so far I've been able to

- Switch back and forth between MF and AF.
- Switch back and forth between Movie and Stills mode.
- Take stills pictures.
- Record movies.
- Query the camera for the lens focus position.
- Set the lens focus position.
- Control the focus speed (how fast the lens changes to a new focus position)

In the screenshots from my logic analyzer below, Async 0 is the camera's RX line, i.e. commands from my computer to the camera. Async 1 is the camera's TX line, responses from the camera to my computer.

If i send this, which is the **GET** CAMERA_CONFIG_LENS_FOCUS_POSITION command,
[image]
the camera responds with
[image]
The four bytes highlighted in yellow hold the current lens position. In the pic above it's focused at infinity. When focused at its near limit, the camera's response on the query  is
[image]
Note the change in the yellow framed bytes. I have only tested with one lens (Olympus 25mm/1.8) so far, but I suspect the position values for close focus and infinity might be different for other lenses.

Finally, by sending **SET** CAMERA_CONFIG_LENS_FOCUS_POSITION commands, it is possible to position the focus point anywhere between the end points.

Command to set focus to near limit - focus position in the four bytes framed in yellow:
[image]

Command to set focus to infinity:
[image]
Command to set focus somewhere in-between:
[image]

I'm still very much in exploratory mode. I feel that the documentation on the UART commands is somewhat lacking in details... :stuck_out_tongue_winking_eye:
So, I've been doing a lot of trying out things, testing and experimenting. But at least I now know the key press sequence to reset the camera to factory defaults by heart :laughing:

I'll be away for a while, but I'll try to do a little write-up and maybe some code example when I get back.

Cheers, Ragnar

MechtEngineer, 2018-05-09:
@garciaanto would you mind posting some simple example Arduino example code using the commands from this Github "Z-Camera-Doc/expansion/expansion.md"? For example, changing from the default video mode to stills mode? 

If I can connect to the Z-CAM via I2C without having to use an EEPROM, that would be amazing!

MechtEngineer, 2018-05-11:
@RagnarJensen I noticed that in your focus commands, the third byte is 0x07, whereas the UART documentation says the third byte in UART SET CONFIG commands should be 0x04. Is there any reason for you changing it to 0x07?

RagnarJensen, 2018-05-11:
@MechtEngineer The third byte is the length of the command or the response. E.g. in the GET CAMERA_CONFIG_LENS_FOCUS_POSITION command it is 0x02 and in its response it is 0x13.

I have no idea what the **set MF drive** command is supposed to do. I've only looked at it briefly, but I think I saw the focus position in the responses. And in their example,  the length byte of it seems to be wrong.

RagnarJensen, 2018-05-12:
I have actually never written code for an Arduino before... :flushed: This is my very first attempt ever, so go easy on me...
I have a couple of [Teensy 2.0](https://www.pjrc.com/store/teensy.html)s left over from another project. I didn't write any code for them, other people did. I just integrated stuff.

The Teensy is a very close relative to the Arduino, it uses use the ordinary Arduino IDE for coding stuff . So, I thought "how hard can it be?" Yes, famous last words, I know :wink:
It turned out not being too hard after all. I have managed to slap together a proof-of-concept sketch that actually does some things OK.

From the comments in the code:
"What this sketch can do:
 *  Simulate an i2c EEPROM, to make the camera configure itself for UART mode.
 *  At start-up, it waits for the camera to come alive. 
 *  At start-up, it sets the camera to Stills mode.
 *  At start-up, it sets focus mode to AF.
 *  At start-up, it focuses the lens to infinity.
 *  Press button 0 to toggle between MF and AF.
 *  Press button 1 to slew focus between infinity and closest focus. Press again to go in the other direction.
 *  Press button 2 to focus farther away, in a step-wise fashion. There are five steps.
 *  Press button 3 to focus closer, in the same steps as above.
 *  Press button 7 to take a picture.
 *  Blinks the on-board LED in different patterns, to show status."

The code so far is of the "ugly hack" variety. It's definitely not pretty, but it may serve as an example for how to do UART communications with an Arduino.

I have set up a repository where I will publish the results of my experiments.
[RagnarJensen/Z-CAM-E1-IO](https://github.com/RagnarJensen/Z-CAM-E1-IO)
 
[image]
Bitscope oscilloscope/logic analyzer, breadboard, camera, batteries as focus targets and a CVBS monitor.
The grey, furry things on the camera are [Rycote Micro Windjammers](https://rycote.com/microphone-windshield-shock-mount/micro-windjammers/).

--
Ragnar

MechtEngineer, 2018-05-14:
@RagnarJensen Wow! This is absolutely magnificent! Thanks for all of your effort in posting your code and circuit. Outstanding stuff. I salute you!

RagnarJensen, 2018-05-19:
Using flash with the E1. That's the reason I got interested in the API and UART control in the first place.

I've been mucking about with my Arduino code and added a resistor and a transistor to the breadboard. I can now trigger a flash from the controller. There are limitations, though. The electronic shutter's slow read-out speed and delays of varying duration in the camera make for some interesting challenges.

Head over to [my repo](https://github.com/RagnarJensen/Z-CAM-E1-IO/tree/master/Flash) for more info.

[image]

@Chenel000 Thanks for the postcard :smile:

snickb, 2018-06-01:
Has anybody been successful communicating with the e1 with an arduino? I have been unsuccessful trying to replicate Ragnar's results. 
The arduino sends 0x03 when the cam is powered on but is not responding to any commands send through the uart

Chenel000, 2018-06-01:
Nice you like it @ @RagnarJensen

gius19, 2022-08-06:
Hi,
I found these resources on Z cam E1 looking in internet for help.
I am interested in this camera for setting up a group of 4 cams to take pictures at the same time and then  to stitch in pano pictures.
I will use manual fisheye lenses, no need to focus. Before buying I would need to to know if this cam may comply some requisites.
1) I have a sony camera and for this purpose I use the function "enable picture without lens": is it possible to do the same with Zcam E1?
2) Is it possible to set "aperture priority" and set a maximum limit in exposure time (e.g. not more than 1/1000s)?
3) I would like to connect all the cams (via usb or other ports, connecting cables) to take a picture simultaneously. Is there a port for wired shutter release?

Thank you in advance for any hints and help.

Giuseppe