imaginevision/Z-Camera-Doc #27: SBUS Control

geekness, 2018-06-05:
I see in the 'expansion.md' file that SBUS is read on the CVBS pin.
Does it also need ground?

The document also says the signal needs '0x01 in first byte'
How do I know if my SBUS signal has the 0x01 in the first byte? Im thinking of using a pixhawk flight controller's SBUS out port to send the signal to the camera to change mode and do capture and record.

acseyang, 2018-06-06:
actually i also want use this camera to aerial film.i also do't know how to open the cvbs.the documents do’t show the code

RagnarJensen, 2018-06-06:
First off, the pins in the camera's I/O port:
[image]

SBUS is a serial UART protocol and is done on the camera's UART pins 5 and 6. In the "SS01 GPIO TEST.pdf" file, you'll find an SBUS interface circuit example:
[image]

You need an external[ i2c EEPROM](https://www.engineersgarage.com/electronic-components/24c02-eeprom) at i2c address 0x51, connected to the camera's i2c pins 7 and 8 in order to configure the pins of the I/O port.
 When the camera boots, it tries to read byte 0 from an EEPROM at i2c address 0x51. **That is the "first byte" the expansion.md document is talking about, the first byte in the EEPROM**.

If the first byte in the EEPROM is 0, the camera configures its pins 5 and 6 as remote shutter lines. Connecting pin 5 to ground acts as a half-press on an ordinary shutter button (the camera auto focuses), connecting pin 6 to ground fires the shutter.

If the first byte in the EEPROM is 1, the camera configures its pins 5 and 6 for the SBUS protocol.

If the first byte in the EEPROM is 3, the camera configures its pins 5 and 6 for asynchronous serial UART communication.

CVBS (composite video output) is always enabled on pin 10, if the camera successfully reads the external EEPROM. That's why they write _"CVBS + whatever..."_ in expansion.md. Pin 10 is not used for anything else.

Instead of an EEPROM, you can use an Arduino. You connect the Arduino's i2c lines to the camera's and load this sketch:
```
#include <Wire.h>

void setup() {
  Serial.begin (9600);
  Wire.begin(0x51);
  Wire.onRequest(requestEvent);
}

void loop() {
  // :DD
}

void requestEvent() {
  Wire.write(0x01);
  Serial.println ("cvbs+sbus");
}
```
When the camera makes its read request to address 0x51 on the i2c bus, the Arduino responds with the number 1 and the camera configures itself for SBUS on its pins 5 and 6.

--
Ragnar

geekness, 2018-06-07:
Thankyou so much for your detailed and concise response.
Has anybody ever taken the time to fabricate a circuit board as per the "SS01 GPIO TEST.pdf" file?

RagnarJensen, 2018-06-07:
[image]
A test circuit. Just LEDs connected to all active pins of the I/O port. Watch the blinking lights when there is activity on the pins.

[image]
An example of electrostatic discharge protection. Resistors in series with all data lines and [ESD suppressors](https://en.wikipedia.org/wiki/Transient-voltage-suppression_diode) (D8-D14) to clamp high voltages down.

[image]
A 3.3 volt regulator.

[image]
The external EEPROM. Pins 5 and 6 of this would connect to camera's pins 7 and 8.
R26, R30 and R31 are 1 kΩ. R27, R28 and R29 are 10 kΩ. This causes address pin A0 to be pulled up close to VCC, while A1 and A2 are pulled down towards ground. This changes the chip's default i2c address of 0x50 to 0x51. The picture shows it powered from 3.3V, but it works fine on 5V, too.

[image]
No idea. Some connector, somewhere inside the camera...?

And yes, you need a common ground for all connected devices.
https://forum.arduino.cc/index.php?topic=99708.0

Cheers, Ragnar

acseyang, 2018-06-08:
Can we use UART to control the camera through the RC's PWM signal?
Like if PWM>1600  it can trigger the shutter button and PWM<1500  will trigger the mode  switch and PWM around 1500 do nothing.That we only need one channel to control the camera's mode switch and shutter.And use the rc‘s knob we can adjust the camera’s shutter speed or EV.so that we don't need that extra s.bus module.
ps:And i saw you used a pwm signal to control the lens

RagnarJensen, 2018-06-08:
Yes, that is certainly doable. It's easy to determine the PWM pulse length with an arduino.
http://www.benripley.com/diy/arduino/three-ways-to-read-a-pwm-signal-with-arduino/

P.S. No, I don't use a PWM signal to control the lens, only buttons and an analog potentiometer.

RagnarJensen, 2018-06-17:
@acseyang 
I have done some quick tests with PWM from an RC receiver. It definitely works 😄 
Head over to [the sketch](https://github.com/RagnarJensen/Z-CAM-E1-IO/tree/master/RC_PWM) for some inspiration.

Cheers, Ragnar

geekness, 2018-06-26:
@RagnarJensen 
There are a number of us that are interested in building a board/connector for the IO port, that can either allow SBUS or UART control of the E1, likely with a solder bridge or dip switch or something.
Is the SBUS schematic above the same schematic for UART control? Would we just have to get the eeprom to output 0x03 or 0x01 as required?

RagnarJensen, 2018-06-26:
_There are a number of us that are interested in building a board/connector for the IO port, that can either allow SBUS or UART control of the E1, likely selectable by a solder bridge or dip switch or something._
:sunglasses: Yup, I've googled around a bit and it seems there is a quite a lot more interest than what I first thought.

_Is the SBUS schematic above the same schematic for UART control? Would we just have to get the eeprom to output 0x03 or 0x01 as required?_
No, they are quite different. SBUS is a single-wire protocol, all devices are connected to one and the same data wire, i.e they are all sitting on the same data bus, hence the name S**BUS**. It is a packet oriented protocol, the RC receiver sends a 25-byte packet every 14 ms (7 ms in high speed mode) at a speed of 100000 bits per second.

Asynchronous UART is a point-to-point connection, i.e. exactly two devices talking to each other. It's character oriented, so there's no data packets. Data is sent one byte at a time. It uses two wires, one in each direction. The transmitting side of device A (TX) is connected to the receiving side on device B (RX) and vice versa. The speed can be anything from 50 bps (or even lower) to several Mbps. The Z CAM E1 operates at 115200 bps. Also, the voltage levels representing ones and zeroes are inverted compared to SBUS. That's why there's a SN74AHC1G04 chip, it inverts the signal from the RC receiver, to something the electronics in the camera can understand.

_Im not really knowledgeable on EEPROMs or coding etc, so i'll ask some questions:_

_1. Is the address set by how the EEPROM is wired?_
Yes, the chip has three address pins A0 - A2, that you tie to Vcc or ground, giving eight possible  addresses between 0x50 and 0x57.

_2. To get the EEPROM to output a byte, it needs some code uploaded to it?_
It needs to be programmed with data. Only the first byte of it needs to be programmed. I use a Raspberry Pi, which has an i2c bus, and put the EEPROM chip on a small breakout board like this one:
[image]

_3. In the SBUS schematic above, which connections go to the i2c pins on the camera?_
In that schematic, there's only SBUS. The i2c connections are not shown at all.

_4. In the SBUS schematic above, What do the Rx and Tx pins do? Are they the SBUS signal, that's coming from the connector on the far right?_
Yes, the connector on the right represents the SBUS output on RC receiver. Rx and Tx are the camera's pins 5 and 6.

I've used my mad Photoshop skillz :stuck_out_tongue_winking_eye: to paste some of the pics together and draw some lines, to hopefully show how the SBUS connections fit together.
[image]

--
Ragnar

geekness, 2019-02-22:
@RagnarJensen 
Im going to put a schematic together on "EasyEDA", and then have a PCB manufactured for this schematic you show above.
If I get it together, are you able to help review and help with programming?

geekness, 2019-02-22:
Actually, one question before I get into that, what functions can be controlled on the E1 with SBUS?

RagnarJensen, 2019-02-23:
@geekness 
In the [expansion.md](https://github.com/imaginevision/Z-Camera-Doc/blob/master/E1/expansion/expansion.md#sbus) document, at the very bottom it says

> SBUS
> channel 4 - mode switching
> channel 5 - capure/rec
> 
As I understand it, the only functions available on the SBUS interface are Toggle Video/Stills mode on channel 4  and Take Picture/Record Video on channel 5.

I'll be  glad to help if I can, but I don't own any SBUS RC hardware, so I won't be able to test anything.

geekness, 2019-02-25:
hmm, maybe i've misunderstood the functionality.
What I was thinking is that I could convert a SBUS signal into the uart control through that last schematic.

But now I realise that needs an arduino. Not sure what I was thinking last week. 

I'll check out the airbot thingy and see what they are doing.

K3FF, 2020-02-25:
Hi guys, if you just want to get the CVBS output ON from your drone, you can use the existing Z-Cam AV interface : https://www.airbot-systems.com/produit/module-retour-video-av-interface-z-cam-e1/?lang=en
Airbot Systems also made an RC control interface : https://www.youtube.com/watch?v=J5smrrNqGsY

Best

scstudios1, 2020-12-21:
Hi, does anyone know of a way to get the plugs so I can use the IO port on the camera?  I have seen all the old links and they are all dead.  I just got two E1 cameras from izugar and thats all fine, but i need a right angle adapter so I can easily mount these cameras next to each other without the giant stick poking out of the side of the camera.

K3FF, 2020-12-21:
The Airbot Systems remote control interface is out now (https://www.airbot-systems.com/produit/z-cam-e1-rc-control-interface/)
Anyway if you only need the Z-Cam E1 usb plugs, you can contact them directly

scstudios1, 2020-12-27:
I contacted airbot, they said contact zcam for the plugs.  Did that, they said contact izugar, did that, he said, i dont know.  I looked online and see that this 12 pin plug is rare, it was used as the usb port on the Casio TRYX camera.  You can still buy cables for that port, but i assume they dont use all 12 pins, so that sort of wont work unless you tear the cable apart and try to solder up all the pins needed.  I see female plugs available on ebay that are 12 pin and that looks fine.  12 pin male plugs are also on ebay, but the picture they have is 14 pins, so I ordered those and will hope they really are 12 pins as the description says they are 12 pin.  Even after contacting that ebay seller, asking which was right, the picture or the description, they replied, we are selling whats on ebay, yeah, great.  My biggest problem is simply getting these plugs as i have working electronics for what i need, i just need to build a right angle adapter for that 12 pin.   So, in the end, my biggest roadblock is the plugs.  I have a uart sync solution that works great from iZugar.  Its just that i cannot deal with these big 12 pin usb dongles hanging out of the camera.  Hopefully, these plugs i ordered are 12 pins, if not, this thing is doa.

K3FF, 2020-12-27:
Lol, i work at Airbot and i don't remember having a request for connectors. Anyway, we have hundreds of them, so do not hesitate to contact by using our website contact form  : https://www.airbot-systems.com/contact/
Merry Christmas !

scstudios1, 2020-12-28:
Really,Remember this email from you waaaaaay back on dec 21?
K3FF <notifications@github.com>UnsubscribeTo:imaginevision/Z-Camera-DocCc:scstudios1,CommentMon, Dec 21 at 12:52 AM

The Airbot Systems remote control interface is out now (https://www.airbot-systems.com/produit/z-cam-e1-rc-control-interface/)
Anyway if you only need the Z-Cam E1 usb plugs, you can contact them directly

scstudios1, 2020-12-28:
The form to ask a question on your website does not work by the way.  You fill it out and it says there is an error.

K3FF, 2020-12-28:
The contact form works fine from our side but we are interested to get a screen shot or something that could help to see what happened. Anyway you can contact us with the following email : support([at])airbot-systems.com
Simply replace the signs by @. Best !