imaginevision/Z-Camera-Doc #29: Advice needed on getting started with http

amati8, 2019-01-31:
Hi all. 
I have an E2 which i'm trying to communicate with in Direct Ethernet Mode.

I'm completely new to http so am using Restlet Client in my Chrome browser (on Win7). I'm using the GET method and sending http://10.98.32.1:80/info but getting zero response.
Am I doing this totally wrong?
Any help would be greatly appreciated!

Charlie.

Christoph460, 2019-02-04:
@amati8  Do you have already gained some insights how to do so? Would be interested as well. Thanks for sharing.

amati8, 2019-02-05:
No, not yet Christoph460.
I'm still waiting for help from someone here.

ailike, 2019-02-12:
check if it work with your browser.

amati8, 2019-02-15:
It works with a browser using Network Mode set to Direct, and typing http://10.98.32.1/www/index.html

It does NOT work with Network Mode set to Router and connecting to PC via my Gigabit Router. The LEDs on the  Z CAM Ethernet Port flash but the ETH. IP does not show an assigned number.
I need to use a Router because I have two cameras to control.

Z CAM running on firmware version 0.82

ailike, 2019-02-21:
Set the camera to Router mode, then shutdown your camera. 
Connect the Ethernet cable to your routner, and boot up the E2. It should work.

If not, please check the router setting and the DHCP client list inside the router.

amati8, 2019-03-20:
Thank you Ailike, I can now control the E2 via my router using a browser via the Controller.html

Is there any way to stop the E2 streaming video to the browser?

I ask this because I need to control two or more E2s using a browser via a network wireless link, and want to reduce the bandwidth required.
Z CAM running on firmware version 0.84

ailike, 2019-10-12:
write your own page:)

closed 2019-10-12 by ailike