imaginevision/Z-Camera-Doc #34: CORS headers

qhulamm, 2019-04-30:
I've built a web application to control the zcam e2 for my project.  I'm having issues running it though because the camera does not send the CORS headers which does not allow my application to access/control the camera.  Can this be added?

ailike, 2019-05-10:
Noted

cscarfie, 2019-07-22:
Has there been any resolution to the CORS policy issue?  I am building a browser based client for the E2C and am running into the same problem.  .htaccess file is missing on the camera to allow wildcard CORS access.  Any way to load up a custom .htaccess file?

rlamarche, 2019-07-23:
A solution could be to run a proxy that send the CORS headers.

ailike, 2019-10-12:
Already supported in latest release.

closed 2019-10-12 by ailike