imaginevision/Z-Camera-Doc #42: Streaming Latency

xqms, 2020-05-20:
Does anybody know what kind of latency I can expect from the builtin network streaming?

It seems there are different methods for streaming (RTMP, SSP).

@q3o Since you recently asked about capture parameters, do you know something about this?

q3o, 2020-05-20:
unfortunately no, I comunicate with it via http but I haven't tried streaming yet. I might give it a go this weekend althou I wont be the best judge because it's running over 4G so it will have some added lag

xqms, 2020-05-20:
Wow, thanks, I'd appreciate any data on this!

However I'm looking for latencies <100ms (preferably <50ms), so if you feel your data connection already has more latency than that it might not be worth it ;)

rlamarche, 2020-05-24:
I'm also interested to know the latency, and if it is possible to serve the stream as webrtc. Thanks.

huaerduoduokaibuting, 2020-06-02:
Does everyone use http to pull 4k stream? I see that about http protocol only how to set the parameters of stream0/1, but did not give a method of how to use http to pull the stream.

hhaidar, 2020-08-15:
Yes, would be good to get some clarification on how to pull 4k

ananords, 2021-05-07:
Hi! Did anyone figure out the latency when network streaming?

LiGoldragon, 2023-07-16:
There seems to be no information on streaming protocols supported, latency, ssl support, etc. Is the plan to boot an open OS on it? The progress right now is from all fronts, so AI and web3 are going to interact with the chips driving the sensors everywhere. Be two steps ahead. I'd be happy to do a free short consult with you. Thank You :pray:

filiphanes, 2025-04-23:
I my use case ZCAM E2 via LAN cable via 1Gbit switch to OBS have latency about 4 frames 1080@30fps it is about 120ms from glass to glass (monitor). Using RTSP stream to obs-streamer plugin, because included media source has higher latency.

SSP native protocol with obs ssp plugin has lowest latency maybe on par with RTSP. I am not using it because ssp plugin was not stable for me.

SRT has slightly higher latency maybe 7 frames ~ 200ms and more dropped frames.
Nick Nordquist in FB group said about SRT: _Latency 50ms to 5000ms (typical is 150ms) if caller and listener latency don’t match, whichever is the longest is the default._