imaginevision/Z-Camera-Doc #45: RTSP pull stream can only pull to 14.99FPS？

HongShiKuan, 2020-06-04:
The stream1 has been set to 4K, 30FPS through the HTTP protocol. But why the stream that I pull through RTSP is only 4K, 14.99FPS?
The following is the information I checked the RTSP stream through FFmpeg and found that FPS=14.99, tbr=29.97.
Has anyone encountered this kind of problem? thank you very much.
`Input #0, rtsp, from 'rtsp://192.168.1.101/live_stream':
  Metadata:
    title           : Session streamed by "OnDemandRTSPServer"
    comment         : live_stream
  Duration: N/A, start: 0.000000, bitrate: N/A
    Stream #0:0: Video: h264 (Baseline), yuv420p(tv, bt709, progressive), 3840x2160, 14.99 fps, 29.97 tbr, 90k tbn, 29.97 tbc
    Stream #0:1: Audio: aac (LC), 48000 Hz, stereo, fltp`

ailike, 2020-06-22:
It's a bug with 0.95, the fps is correct but in the H.264 meta data it's half

ailike, 2020-07-02:
Fixed in 0.96.0

closed 2020-07-02 by ailike