imaginevision/Z-Camera-Doc #52: Is Pelco-D on CTRL is RS232?

IvanDev, 2020-08-15:
I'm planning to connect Arduino to CTRL port of Z CAM E2-F6 in Pelco-D mode to process pelco commands from ZCAM, unfortunately I couldn't find any documentation on CTRL in Pelco-D mode. 
Does it work in RS485 as expected by Pelco-D standard (in this case which tx/rx is RS485 A/B)? Or in RS232 as stated by camera documentation?
Can CTRL 5V line handle Arduino (300-500mA max)?

momosh13, 2020-11-03:
Здравствуйте, 
Я вижу что ни кто не отвечает на ваши вопросы. У меня кое что получилось. Я подключил Arduino к ZCAM. Если вам ещё нужна помощь то я могу отправить вам схему как я подключил Arduino и могу дать программу, но у меня не получилось найти все команды которые мне были нужны. Если вы думаете что можете найти все команды мы можем вместе попробовать завершить наш проект. Вы можете написать мне на мой email или здесь. 
Мой email momosh66@hotmail.com

momosh13, 2020-11-03:
> I'm planning to connect Arduino to CTRL port of Z CAM E2-F6 in Pelco-D mode to process pelco commands from ZCAM, unfortunately I couldn't find any documentation on CTRL in Pelco-D mode.
> Does it work [...]

Здравствуйте, 
Я вижу что ни кто не отвечает на ваши вопросы. У меня кое что получилось. Я подключил Arduino к ZCAM. Если вам ещё нужна помощь то я могу отправить вам схему как я подключил Arduino и могу дать программу, но у меня не получилось найти все команды которые мне были нужны. Если вы думаете что можете найти все команды мы можем вместе попробовать завершить наш проект. Вы можете написать мне на мой email или здесь. 
Мой email momosh66@hotmail.com

ailike, 2020-11-05:
> I'm planning to connect Arduino to CTRL port of Z CAM E2-F6 in Pelco-D mode to process pelco commands from ZCAM, unfortunately I couldn't find any documentation on CTRL in Pelco-D mode.
> Does it work [...]

400mA is OK

momosh13, 2020-11-05:
> 400mA is OK

Hi ailike
I able to figure out some commands (RS232) to control ZCAM with Arduino,
This is the command:
ISO+
ISO-
IRIS+
IRIS-
SHUTTER+
SHUTTER-
REC START
REC STOP
I need couple more commands to compete my project, if you can, please help me,
I need:
FOCUS+
FOCUS-
WB+
WB-
Thank you.

Deca, 2024-10-21:
> Hi ailike I able to figure out some commands (RS232) to control ZCAM with Arduino, This is the command: ISO+ ISO- IRIS+ IRIS- SHUTTER+ SHUTTER- REC START REC STOP I need couple more commands to [...]

Hi @momosh13  have you managed to make progress on adding the focus command? Also have some code to share in order to to control the ZCAM with Arduino via serial CTRL
The documentation about that is basically non-existant, I've managed to connect the arduino via the serial port in uart mode and I see data on the serialmonitor. Then I tried to send the byte for retrieving ISO configuration with the getConfig request ( passing  EA 02 03 2E 00 02) but from this point I don't really know what to look as a response. 
It took me days just to understand that I needed to look in uart_def.h for the correct enum value to pass :(
Any help is appreciacted!