imaginevision/Z-Camera-Doc #61: get lens_focus_pos key value

adahbingee, 2022-05-17:
Please add the key value.

```
GET /ctrl/get?lens_focus_pos
```

[image]

costavale, 2022-07-24:
Did you manage to understand what are supposed to be the values? and if the values are the same as mf_drive?

adahbingee, 2022-07-25:
It' works

command to get focus position
```
/ctrl/get?k=lens_focus_pos
```

for setting focus command
```
/ctrl/set?lens_focus_pos=100
```

closed 2022-07-25 by adahbingee

costavale, 2022-07-26:
Thank you! I have seen that this is working, but what does the number "100" in `/ctrl/set?lens_focus_pos=100` means? and what is the difference to `/ctrl/set?mf_drive=100`

adahbingee, 2022-07-26:
sorry, that's just an experienced value

costavale, 2022-07-26:
Ok, and what is the difference with mf_drive?

adahbingee, 2022-07-27:
I did not use this command yet.
sorry

costavale, 2022-07-27:
Ok, thank you!