imaginevision/Z-Camera-Doc #69: E2 F6 HDMI out, no 60fps in 4K , despite the specs

nazoren, 2024-09-01:
Hello,
Please we need help, trying to get 60fps in 4K resolution from HDMI out of E2 F6 with firmware v1.0.3.
We are trying to achieve the smooth 4K60p HDR streaming to Youtube.
Capture card is Blackmagic Design DeckLink Quad HDMI Recorder with latest drivers.
Sony ZV-E1 camera we tested with the same settings and works fine every time:
https://helpguide.sony.net/ilc/2310/v1/en/contents/201h_HDMI_output_setting.html

Couldn't achieve this with E2 F6 although specs on the website reads:
https://www.z-cam.com/e2-f6/
HDMI 2.0 Type A (video output up to C4K @ 60fps, 10-bit 4:2:2)

By the way we only need 2160p 60fps with 10-bit 4:2:0 due to Youtube requirements:
https://support.google.com/youtube/answer/10265272?hl=en#zippy=%2Cobs

Below is the test video, showing Sony works fine but Z Cam downgrades the resolution to 1080 if we try 60fps:
https://www.youtube.com/watch?v=OCkYbMQ5QWY

Thank you!

nazoren, 2024-09-02:
Got the instructions from Z cam support, will test and update. Thank you!
For future reference:
About HDMI output, please clear settings in the menu of the camera, then set the project to C4K59.94, then you will find C4K60 in the menu of the HDMI (closed EDID). I don't know what were the settings you used, so, clear settings will be a quick way.

maxholgasson, 2025-06-19:
> Got the instructions from Z cam support, will test and update. Thank you! For future reference: About HDMI output, please clear settings in the menu of the camera, then set the project to C4K59.94, [...]

how did you contact them as they dont reply to issues here