imaginevision/Z-Camera-Doc #72: Changing WB and tint with UART command

maxholgasson, 2025-03-07:
Hi,
is there any documentation how to set WB and tint with an UART command for the E2 M4?

milliyang, 2025-12-12:
a sample code for all uart command control:
   https://github.com/imaginevision/Z-Camera-Doc/blob/master/E2/protocol/uart/example/main.cpp

milliyang, 2025-12-15:
You are an experienced developer,  just take a look at the simple code.

you should do the followings:
  1. get_config(key=wb)
  2. set_config(key=wb, value=your_value)

for example, (in http):
  1. http://{ip}/ctrl/get?k=wb
  2. http://{ip}/ctrl/set?wb=Manual
  3. http://{ip}/ctrl/get?k=mwb
  4. http://{ip}/ctrl/set?mwb=5000
  5. http://{ip}/ctrl/get?k=tint
  6. http://{ip}/ctrl/set?tint=0

so, send the sample uart command just like http.

maxholgasson, 2025-12-16:
> => Do you means the Web Controller, "Image Setting Tab".

Yes, amazing options for future yes, so far I'm happy with the current way but might be useful for future applications.

Indeed I'm now continuing with a http solution. I needed to get a live iso value and over serial I got only "auto" when in auto-mode whereas via http I get the current iso value the camera auto selected.
Ordered now some dev boards with ethernet.
I prefered UART as I need a very low power device but seems there aren't just enough options with UART